LumaTeam/Luma3DS #1907: Can you cfw an emulator?

SuperMarioYAHOOYAHOO, 2023-06-14:
I want to install Luma 3DS on CItra. im not sure if you can, as i seee no guides on how to do it. is this possible?

Shadow322, 2023-06-14:
> I want to install Luma 3DS on CItra. im not sure if you can, as i seee no guides on how to do it. is this possible?

Take this to gbatemp.net this is not a help forum

ihaveamac, 2023-06-14:
Citra is not a low-level 3DS emulator so it doesn't run ARM9 software like Luma3DS or GodMode9

SuperMarioYAHOOYAHOO, 2023-06-14:
> > I want to install Luma 3DS on CItra. im not sure if you can, as i seee no guides on how to do it. is this possible?
> 
> Take this to gbatemp.net this is not a help forum

bruh i did

SuperMarioYAHOOYAHOO, 2023-06-14:
> > I want to install Luma 3DS on CItra. im not sure if you can, as i seee no guides on how to do it. is this possible?
> 
> Take this to gbatemp.net this is not a help forum

https://gbatemp.net/threads/cfw-on-emulator.634648/#post-10183992

Masamune3210, 2023-06-16:
You were given an answer in the thread, yet you persevered and decided to make a issue to ask here as well? Citra cannot run Luma, it is not a low-level emulator.

SuperMarioYAHOOYAHOO, 2023-06-17:
> You were given an answer in the thread, yet you persevered and decided to make a issue to ask here as well? Citra cannot run Luma, it is not a low-level emulator.

bruh i made this before i got the answer

closed 2023-06-17 by SuperMarioYAHOOYAHOO