LumaTeam/Luma3DS #1910: [Feature request] Make Luma3DS childproof using config.ini

erikiorai, 2023-06-18:
Ok so i want to mod a 6 year old Nintendo 3DS but i'm afraid that after doing that later it might break it accidentally. So it wil be better if you have the option to disable luma configuration menu,rosalina menu,luma chainloader inside `config.ini` like this.
```
luma_configuration_options = false
rosalina_menu = false
luma_chainloader = false
```
Especialy important to have this feature in chainloader and in rosallina because those has full controll of the 3DS system. Especially godmode9, now there is button combo safty but you know kids are kids and it wil press those buttons godmode9 gives blindly

erikiorai, 2023-06-20:
Well i gived a rooted phone to my granpa and he broke it (softbrick) so thats why im afreid

Masamune3210, 2023-06-20:
Its a lot easier to brick a rooted phone than it is to brick a 3DS. Just don't keep anything on the card that would be dangerous. You cant use godmode to brick the 3DS if godmode isn't on the sd card

PabloMK7, 2023-07-17:
Hello, this feature seems to be very specific and there are other ways to solve it. For example, you don't need to have godmode9 sitting in the payloads folder, you can remove it (or rename it to something else not ended with `.firm`) so that it is not accessible. I cannot think of any other functionality that may be problematic if activated accidentally.

If we implemented such feature, there could be cases where people more easily fake not having CFW installed, as opening the luma config screen is the easiest way to check, or people changing the configuration and then forgetting how to change it back and would make assistance more difficult.

closed 2023-07-17 by PabloMK7