LumaTeam/Luma3DS #1911: triforce heroes (PAL) crashes when loading download play in US system

F3arM3, 2023-06-21:
**System model:**

New 3DS XL

**11.16.0-49U Sysnand(+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[boot9strap]

**Luma3DS version:**

12.0.1

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: (off )

Splash duration: ( Unknown)

PIN lock: (off )

New 3DS CPU: ( off)

--

Autoboot EmuNAND: (N/A )

Use EmuNAND FIRM if booting with R: (N/A )

Enable loading external FIRMs and modules: (No )

Enable game patching: ( No)

Show NAND or user string in System Settings: (Yes )

Show GBA boot screen in patched AGB_FIRM: (No )

Set developer UNITINFO: (N/A )

Disable Arm11 exception handlers: ( N/A)

Enable Rosalina on SAFE_FIRM: (N/A )

--

**Explanation of the issue:**

when I tried to boot download play my system crashed

**Steps to reproduce:**

1.play triforce heroes PAL on modded us new 3ds xl

2.walk into multiplayer lobby and talk to old man, click download play and create team

**Dump file:**

It does not list a dump location so I will transcribe it
"An error occurred (ErrDisp)
Error type: generic
Process ID: 38
Process name: dlp
Process title ID: 0004013000002802
Address: 0x0010440e
Error code: 0xc8a11bef
Press any button to reboot.

Sorry if this  isn't a luma issue, if it isn't one please let me know where I should submit it and close this issue

PabloMK7, 2023-07-16:
It is possible that the out of region game may be causing issues. Ideally, you should try to get a game that matches both console regions.

rdistiso, 2023-10-19:
Try updating your setup and use Luma Locale switcher

closed 2024-03-30 by TuxSH