LumaTeam/Luma3DS #1913: [Feature request] volume control

Anonymous941, 2023-06-25:
I have an old 2DS, and the volume slider is broken, and it appears I'm [not](https://gbatemp.net/threads/nintendo-2ds-volume-slider-broken-and-stuck-on-the-loudest-settings.382529/) [the](https://gbatemp.net/threads/2ds-volume-slider-repair.605898/) [only](https://gbatemp.net/threads/n2ds-xl-volume-slider-repair.531757/) [one](https://www.ifixit.com/Answers/View/691832/How+would+I+fix+a+volume+button).  My first thought was that this isn't software-controlled, but when using nds-bootstrap, I'm able to adjust it.  Also when I move the console too much, the volume goes back to 83% (where the slider is stuck) until I go into the menu and change it again.

It appears that volume control is in `retail/cardenginei/arm9_igm/source/inGameMenu.c`:

```
case OPTIONS_VOLUME:
{                                                                                                                                                                                                        
    u8 volume = (u8)(sharedAddr[6] >> 16);
    if(KEYS & KEY_LEFT && volume > 0) 
        volume--;
        else if (KEYS & KEY_RIGHT && volume < 31)
            volume++;

            sharedAddr[0] = volume;
            sharedAddr[4] = 0x554C4F56; // VOLU
            while(sharedAddr[4] == 0x554C4F56) {
                while (REG_VCOUNT != 191) swiDelay(100);
                while (REG_VCOUNT == 191) swiDelay(100);
        }
        break;
}

Can this be added to the Rosalina menu, possibly with the option to periodically re-change it?

Anonymous941, 2023-06-25:
I found [an outdated fork](https://github.com/LumaTeam/Luma3DS/commit/dc636d82492d1e87eb51785fa7f2a98617e7ece9) that has this feature, so it can be added fairly easily...

ShelterPush, 2023-07-20:
I am uncertain if it warrants the creation of another feature request, but in addition to changing the volume level, could this feature include the ability to change the number of channels (mono - stereo - surround)? From what I understand of it, the reference commit only controls volume itself.

I frequently use a single earbud for games, but I would like to be able to switch them to stereo or surround when I take out the earbuds without having to shut down the game I'm playing.

cooolgamer, 2023-08-01:
> I found [an outdated fork](https://github.com/LumaTeam/Luma3DS/commit/dc636d82492d1e87eb51785fa7f2a98617e7ece9) ([commit 2](https://github.com/LumaTeam/Luma3DS/commit/a9b05f35a9c0c3f45abab2f0b57a39309461a8fe)) that has this feature, so it can be added fairly easily...

I added that to my fork and it works fine on the latest luma version, you just have to change some things because just copy/paste it doesn't work. So yes, it is possible to have this on the latest luma version

Anonymous941, 2023-08-02:
@cooolgamer Mind putting it in a PR?

cooolgamer, 2023-08-10:
I did not made it, but I put it on luma v13, I will make a pull request with this when I can (not at home rn) and ofc crediting the original creators (Sono and Nutez)

profi200, 2023-08-10:
Fwiw the MCU is not a good place to override volume. If the slider is wonky these regs will not work right. On CODEC however you have full software control via the shutter sound mechanism which is just overriding the slider temporarily.

Anonymous941, 2023-08-11:
What would happen when the shutter sound triggered?  Would it revert back to the previous volume once it's done?

profi200, 2023-08-11:
I have been experimenting with this a bit and got it to work. Shutter sound only happens via I2S2 which is not responsible for DSP sound. I2S1 volume override is not touched by codec module as far as i can tell so the camera app should do what it always does without affecting this setting.

profi200, 2023-08-12:
Here is the proper software volume control code. It sticks even in sleep mode. It does not survive reboots so it must be set each boot. Camera shutter sound is unaffected and the shutter mechanism used in codec module does also not affect this volume override.

By trial and error i found out the maximum volume slider position maps to volume value -20 (-10 dB). I highly recommend not going above this value because i don't know if this will overload the headphone/speaker drivers.

```
s8 volumes[2] = {-64, -64}; // Volume in 0.5 dB steps. -128 (muted) to 48. Do not go above -20.
u8 tmp;
cdcChkInit();
CDCCHK_WriteRegisters2(0, 65, volumes, 2); // CDC_REG_DAC_L_VOLUME_CTRL, CDC_REG_DAC_R_VOLUME_CTRL
CDCCHK_ReadRegisters2(0, 116, &tmp, 1); // CDC_REG_VOL_MICDET_PIN_SAR_ADC
tmp &= ~0x80; // Control volume via register. Do "tmp |= 0x80" to set volume control back to slider.
CDCCHK_WriteRegisters2(0, 116, &tmp, 1);
cdcChkExit();
```

gannon345, 2023-11-16:
Do we have any idea if/when this might be implemented? The 2ds xl in particular was built with a really flimsy volume adjustment. It's very easy to break when doing a battery replacement (ask me how I know 😅) so this would probably be a useful addition.

Anonymous941, 2023-11-17:
I've tried to implement this a while ago but I don't know a lot about C so it was very difficult for me.  I can try again, but it should be easy to implement for someone who knows more about it...

PabloMK7, 2023-11-17:
Currently implementing @profi200 solution.

Anonymous941, 2024-03-06:
Fixed in #1973

closed 2024-03-06 by Anonymous941