LumaTeam/Luma3DS #1919: Luma v13.0 NDS(i) Mode broken

Devine3DS, 2023-07-20:
Hello,

Ive updated all my 6 3DS Consoles (NEW,OLD) to the latest Luma 3DS Version v13.0 and now on all the NDS Mode is broken.
Cartridges blackscreen after launching and TwilightMenu++ works until you load a game then the same happens.

Devine3DS, 2023-07-20:
Okay can be closed its fixed it was because i had a copy of TwBg laying in there

JORGETECH, 2023-07-21:
This is caused by the latest Luma3DS version that added support for changing the scaling matrices in a similar way TWPatcher did (it also has the "unlocked DPad patch"), I guess the old method is incompatible and TWPatcher needs to be updated. The latest Luma3DS changelog explains the change and you can convert the old syntax to the new one with some Python code, sadly, there is no Widescreen patch and I don't know if RTCom patch has been implemented. If you want those features back I suggest downgrading to an old version of Luma3DS, the same thing I'll do since I also use those patches.

commit f40a1cc4 (2023-07-21): Fix external module loading for firm modules such as TwlBg

Closes #1919 and #1921
closed 2023-07-21 by TuxSH