LumaTeam/Luma3DS #1920: Luma3DS 11.13 crash

Gobl943, 2023-07-21:
For a time, I tried booting into my 3DS after installing the new Luma3DS version and it would keep crashing with an Arm11 exception. It would do this when I enable "external FIRMS and modules" setting in the Luma3DS start up menu. Turning it off fixed it but enabling it always crashes consistently.

Shadow322, 2023-07-21:
> For a time, I tried booting into my 3DS after installing the new Luma3DS version and it would keep crashing with an Arm11 exception. It would do this when I enable "external FIRMS and modules" setting in the Luma3DS start up menu. Turning it off fixed it but enabling it always crashes consistently.

Then update your very outdated Luma3DS dude

flanter21, 2023-07-21:
For me, it was because of rehid. Following the instructions here fixed it for me. https://github.com/hax0kartik/rehid/releases/tag/v4.0

TuxSH, 2024-03-24:
> Turning it off fixed it but enabling it always crashes consistently.

More than likely it was due to ReHID

closed 2024-03-24 by TuxSH