LumaTeam/Luma3DS #1922: Luma3DS automatically copies to NAND with consent from user

LilithValentine, 2023-07-21:
Luma3DS now automatically replaces the boot.firm on NAND without the consent of the user. This should come with a prompt asking if the user want to replace their boot.firm and give the option to agree or disagree. Users should not have their system messed with without their consent

TobiX, 2023-07-21:
This feature has been present since over a year (v11.0, June 2022) and AFAIK nothing has changed since the last time this was discussed (#1756)

LilithValentine, 2023-07-21:
The fact that it’s still an issue is a problem. The user should be aware of this and given the option to not have it happen to them.

LilithValentine, 2023-07-21:
Closing since it seem’s Tux’s ego is too big

closed 2023-07-21 by LilithValentine

impeeza, 2023-07-21:
the user will be aware if read the readme.md and documentation what everyone should do. 🙂

fyredragon69, 2023-07-21:
why exactly is it such a big problem to you? the copy luma to ctrnand step has been in the guide for years, i believe it was added to make that step easier.

Masamune3210, 2023-07-21:
Because some people are delirious and think that one file being automatically updated is any different than the vast amount of them that are changed every time they turn the system on....

Wryyyong, 2023-07-28:
Truth be told I also wouldn't mind some sort of toggle for this behaviour (albeit as a compile-time setting, as I fear some end users would unnecessarily screw with the setting if it were based within the config menu) for more advanced use cases such as booting from [fastboot3DS](https://github.com/derrekr/fastboot3DS/), where Luma3DS's auto-copying may be unnecessary depending on your configuration and setup.