LumaTeam/Luma3DS #1923: 3DS XL crash

Freakishink292, 2023-07-22:
When I updated Luma3ds to v13.0, the system wouldnt boot. It could go into godmode9 and the Luma config, but I couldnt get past that. Anyone have a fix? (Model: Nintendo 3DS XL)

pollychii, 2023-07-26:
this happened to me, i just reverted it back to 12.0 and its working fine but idk if it was a freak thing or if it will occur again when i try to update again

TuxSH, 2023-07-27:
Which system versions please?

Also if you're using rehid/wuumibo, please upgrade those too.

pollychii, 2023-07-27:
> Which system versions please?
> 
> Also if you're using rehid/wuumibo, please upgrade those too.

hi, it was 10.07, due to buying an old 3ds xl (not actually mine but i modded it). i also have a new 3dsxl that is on 11.17,can i update safely to 13.0?

Freakishink292, 2023-07-28:
> Which system versions please?
> 
> Also if you're using rehid/wuumibo, please upgrade those too.

I don't use those, the versions are 11.17, and the only thing I'm using is rverse.

closed 2023-08-08 by Freakishink292

Freakishink292, 2023-08-10:
> In case you closed this accidentally my advise is to delete all custom sysmodules you have in "/luma/sysmodules". The newest version is incompatible with some older sysmodules and you will get blackscreens on boot.

I did close it on purpose.