LumaTeam/Luma3DS #1926: C-Stick not working when the Plugin-loader is enabled

Paddyk45, 2023-07-23:
**System model:**
New 3DS
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.17.0-50E
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
boot9strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
v13.0.1
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (x)

Splash duration: (3)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
When playing a game while the default plugin is enabled, the C-Stick does not work, this could also be the case for ZL/ZR

**Steps to reproduce:**

1. Install the default plugin

2. Enable the Plugin-loader in the Rosalina menu

3. Start a game that uses the C-Stick (for example IceStationZ)

4. Try to use the C-Stick, it wont work

**Dump file:**

PabloMK7, 2023-07-23:
Hello, this is a known issue and it's a problem with the default plugin, not with Luma3DS. I'm closing this as unrelated to Luma3DS.

closed 2023-07-23 by PabloMK7

Paddyk45, 2023-07-23:
> Hello, this is a known issue and it's a problem with the default plugin, not with Luma3DS. I'm closing this as unrelated to Luma3DS.

Oh okay, thanks