LumaTeam/Luma3DS #1931: Luma config softlock

cooolgamer, 2023-07-25:
**System model:**

[New 2DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.17.0-50E] (doesn't matter in config anyways)

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[v13.0.1]

**Luma3DS configuration/options:**

The bug is inside this menu

--

**Explanation of the issue:**

The entire console get softlocked and the "x" blink weirdly, this is due to the new "Save and exit" change.

**Steps to reproduce:**

1. Open luma config menu

2. Hold d-pad up or down and press start

PabloMK7, 2023-07-25:
Can reproduce, will take a look. Thanks

PabloMK7, 2023-07-25:
Fixed by f8f69c9

closed 2023-07-25 by PabloMK7