LumaTeam/Luma3DS #1945: Luma3DS v13.0.2 random crash

fenetree, 2023-08-16:
**System model:**
Old 3DS XL

**SysNAND version:**
Sys 11.17.0-50E

**Entrypoint:**
Boot9Strap

**Luma3DS version:**
v13.0.2

**Explanation of the issue:**
Hello,

I'm having random crash really often when playing. It happens on different games. I had it on "Mario & Luigi : Bowser's Inside Story", "Mario & Luigi : Paper Jam Bros" and currently with "New Super Mario Bros. 2".
I sometimes enable some cheats from the rosalina menu. I don't think it's related tho because I enable a cheat and can play for like 1 hour and then crash. Sometimes happens after 2h, 30 mins... and then crash.

It's not always the same message. The last one I just had when writing this post is : "current process bigred 000400000007af00"

I tried deleting and copying again the boot.3dsx and boot.firm files, but didn't fix the issue.

And all the dump files i get are not usable. Maybe I'm not using the correct method tho but can't open any of them. You can see the last one just below.

If anyone have an idea how to fix those random crash that are really annoying cause need to save too often (and some games can't save whenever I want).

Thank you.

**Dump file:**
[crash_dump_00000002.zip](https://github.com/LumaTeam/Luma3DS/files/12364030/crash_dump_00000002.zip)

PabloMK7, 2023-08-17:
Hello,

The problem reported here seems to be related to corrupted storage, please follow the next steps:

If your game is a physical copy or a cartridge, take a look at [this post](https://github.com/skawo/GodMode9-with-Cartridge-Fixer). However, if your game is a digital copy, or if you have installed any updates or DLC to the game, your SD card may be corrupted. The following steps may help:

1) Remove the SD card from your 3DS and insert it into your computer. (Or any other device that is able to read SD cards.)

2) Copy all the contents from your SD card and paste them to a folder in your computer. This step is very important to prevent data loss.

3) Use your preferred formatting tool to format your SD card to FAT32 (even if it is already formatted to FAT32). If you need a guide read [this](https://3ds.hacks.guide/formatting-sd-(windows)).

4) Copy all the files you backed up in step 1 to your SD card again. This will rebuild the file system and remove any possible corruption.

5) Try to replicate the steps where SD corruption was causing issues. If you still are having issues, you will have to reinstall the application as the files were damaged (corruption won't damage them again). Make sure to backup the save file with any save manager before doing the reinstall!

6) If after doing the steps listed above you are still having issues, or corruption happens again in a short time, your SD card may be damaged or unreliable and has to be replaced. You can make sure by following [this guide](https://3ds.hacks.guide/h2testw-(windows)).

If the steps listed here don't help, please consider joining the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh) and asking for help there.

closed 2023-08-17 by PabloMK7

fenetree, 2023-08-17:
Hello,

Thanks for your feedback. I tried your method and played a little today, didn't have crash. It might have solve the problem. Otherwise i keep the discord link in mind. 

Thanks, have a good day.