LumaTeam/Luma3DS #1948: [Feature Request] 800px and Antialiasing toggle

Trickygamers, 2023-08-21:
It would be nice to make things look HD for old and new 3ds's. would have a performance dip but have it be a novelty thing for the old 3ds

PabloMK7, 2023-08-21:
Hello. That's not possible as it requires non-trivial per game patches.

closed 2023-08-21 by PabloMK7

Trickygamers, 2023-08-22:
could you add it to ctgp-7 then?

Trickygamers, 2023-08-22:
thanks but I can't find the support page

PabloMK7, 2023-08-23:
> thanks but I can't find the support page

No need to, I'm the main developer of CTGP-7 and I'll consider taking a look at this at some point.

vengestone-dragoon, 2024-07-21:
I have an idea for implimenting this globally from luma, though i dont have the know-how to actualy impliment it.
if you could force the system to stay in 3d mode when the slider was at the bottom and force specific depth settings, overriding/ignoring what the front cammera sees, i think the system could think its rendering sterioscopic 3d, but the result would be the hd 2d image.

profi200, 2024-07-21:
That's not how it works. The GPU needs to render the whole 240x800 frame. In 3D mode it's configured to render 2 240x400 frames with different angles and 2 separate frame buffers are used for each eye. This boils down to the same answer given above. This requires per-game patches.

vengestone-dragoon, 2024-07-22:
Im saying if you could overide the positions, angles, and distances of the 2 240x400 frames, you could get them to line up as the single 2d view with the hd resolution. Make the two veiws face the same direction with a verry small distance(1px) between them.

vengestone-dragoon, 2024-07-23:
K then,
seems strange to me that nintendo would leave the 3d functionality to developers. Its a feature accrose almost all games and on new3ds its got more features, so i would have thout it would be a system service basicly.
oh well then, patching individual games it is

profi200, 2024-07-23:
My last response since this is not a forum.
An extra service is not required since all we are doing is rendering twice with different settings and using 2 frame buffers. There is nothing really you can hook into except GPU command list submission. This command list can be huge so it's not possible to scan through it and change render parameters without affecting performance. This is the reason they could not fix the gspwn vulnerability because it would impact performance. Even if we did it anyway the game is in control of everything else like buffer allocation.