LumaTeam/Luma3DS #1949: no sound, title , logo or image in my 3ds games. (i thing corruption)

AndrehPosting, 2023-08-22:
I have a problem. I recently bought a new 128gb sd card and I put it in my 3ds I followed your tutorials step by step and everything was fine until the part about downloading the games arrived... I use an external program called Custom installer which is to install cias Much faster directly from the PC. It also happens to me with hshop, ghost eshop and with the fbi and the truth is that I am VERY frustrated... when I download one, two or three games, everything is fine, but when it passes a certain number of games (I don't know exactly why when it has happened it happens in different amounts, approximately 20gb of games) from there my games get corrupted and not just the ones I install but ALL the games I've had on the console. They do not open, the photo or the sound of the menu does not appear when you select the game, the cover does not appear or anything. They are totally black files and when I open them the console crashes and I am very frustrated and I don't know what to do because I have to format my sd card over and over again. my 3ds is Old 3ds XL, I use the guiformater to format it in fat32 and I have the luma updated to the latest version. I hope someone from the comments can help me

AndrehPosting, 2023-08-23:
forget to say that all games are corrupted, absolutely all. The cover is not removed from those that were previously installed, but nevertheless when I select them, the logo does not appear on the upper screen of my 3ds. and when I open them I get an error: an error occurred (ErrDisp)
The SD has been removed
Press any button to restart.

PabloMK7, 2023-08-23:
Hello,

It looks like you are having issues involved pirated content. Therefore, we are not able to help you. Sorry

Also, this is a tool for developers to keep track of issues related to Luma3DS, and not a support forum. You will get better and more fluent support if you ask in a forum related to 3DS homebrew or in the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

closed 2023-08-23 by PabloMK7