LumaTeam/Luma3DS #1950: Data abort - Alignment crash in Yu-Gi-Oh Zexal World Duel Carnival (Jap translation patch)

GamerXYZ2, 2023-08-28:
I run the aforementioned game with a RomFS patch taking the English card files from the English rom and putting them in the Japanese version (this because the English version was severly cut in content). I know it sounds silly to report a Luma3DS crash caused by a fan-made patch, but this used to work fine in older versions of Luma3DS with Arm9loaderhax from years ago (unfortunately I don't know which, as it's been multiple updates ago since I last played the game until now), so I know it's supposed to work. I wouldn't be reporting this otherwise.

When I specifically put the cursor on the card Black Luster Soldier - Envoy of the Evening Twilight, I always get a crash of exception type Data Abort and fault status Alignment. Everything else works fine as far as I know (I of course haven't checked out every single card in the game). For as far as cards in Yu-Gi-Oh went back then, the card has a particularly big effect text (maybe even the biggest in-game?) which could be the problem. I've tried some other cards with big effects but couldn't think of something nearly as big.

I have tried uninstalling the game and taking a fresh new Cia of the Japanese version, as well as a fresh new rom of the English version to extract the English card files from, to confirm it's not some files being degraded over time. Still the exact same issue.

Boot9strap version: first an unknown version, then updated to 1.4 (problem persisted)
Luma3DS version: first an unknown version, then updated to 13.0.2 (problem persisted)
Luma3DS settings: all on default except of course game patching enabled
3DS model: regular 3DS
3DS version: Sys 11.12.0-44E

Error dump (remove the .txt since Github says it doesn't support the file as .dmp): 
[crash_dump_00000000.dmp.txt](https://github.com/LumaTeam/Luma3DS/files/12449630/crash_dump_00000000.dmp.txt)

PabloMK7, 2023-08-29:
Maybe related to #1900 ? You can verify it is a layeredfs issue by recompiling the romfs with the patched files. If that also doesn't work then it's an issue with the patch and not Luma3DS.

GamerXYZ2, 2023-09-02:
It was a rom recompiled with the "patch".
Also, I can now confirm it was indeed about the length. I found out how the descriptions file worked and redid the text in a shorter form, and it works now. No weird symbols were removed (how the file works: hexadecimal digit for a letter, 20 in-between everything, 00 for spaces, and 2E for end of text. All I did was remove some proper hexadecimal digits for letters and of course the corresponding 20's and 00's, and as a result also moved the 2E appropriately. However, it's important to note it's not the 2E that determines how much space is allocated, another file has pointers that determine where texts begin and end for each card, but to my surprise there was no need to touch that file, just shortening the text in the first file was enough)

PabloMK7, 2023-09-02:
In that case, I'm closing this issue as it's not related to Luma3DS

closed 2023-09-02 by PabloMK7