LumaTeam/Luma3DS #1951: All Luma3DS config resets when RTC is reset (e.g. on battery swap)

Devnol, 2023-08-28:
**System model:**
Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.17.0-45E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
b9s

**Luma3DS version:**
v13.0.2 
**Luma3DS configuration/options:**
Screen brightness: (4)

Splash: (Before)

Splash duration: (3000ms)

PIN lock: (6)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
After removing the battery on the console, all Luma3ds settings are reset to default. 
(all settings off except show settings string). This means that the user pin as well as other settings have to be set again from scratch.

**Steps to reproduce:**

1. Change some settings on the menu (e.g. set a pin)

2. Boot to console, all is ok

3. Take out battery, put it back in

4. Boot console, Luma3DS menu appears on its own meaning settings have been reset.

Devnol, 2023-08-29:
After a little testing it seems like pressing power to exit without saving and restarting the console the config will load from sd and all settings will be like you set them, however this still means you can bypass the pin by taking out the battery, which is no big deal but I don't see why if it can be avoided.

PabloMK7, 2023-08-29:
Can replicate. This is because the MCU is reset when the battery is removed, so the luma version number stored in MCU is lost. Spawning @TuxSH

Devnol, 2023-08-29:
That was also my understanding. Is this particularly necessary however? Wouldn't a good solution to that be to check if the config file exists in the sd (or perhaps has some flag set for initial setup complete or whatever) first before trying to reset the whole thing?

Devnol, 2023-09-02:
dOn't YoU lECtUre mE on YouR vOlatIle MeMOry bLaCk MAgic!!11!

I am very much aware of how computer memory works lol. What I reported as a "bug" here isn't the fact that the memory clears, of course that's normal. What I was talking about is how luma3ds resets your config when such a thing happens, or at least tries to.

TuxSH, 2023-09-23:
Sorry for the late reply, this is intended because Luma3DS stores its current version number (as well as current nand location when rebooting between native/lgy modes) so it knows when to upgrade. If this information is invalid, this implies the config.ini entry format might have changed, thus it is cleared too.

You shouldn't leave the battery at 0%, let alone reach it, as this damages the battery.

closed 2023-09-23 by TuxSH

Devnol, 2023-09-23:
well the reason I came accross this wasn't battery drain, but caused by the volatile memory losing power when swapping out batteries. Now this would only occur when either a) opening the system up for repair or b) swapping out an empty battery for a full one, which is actually probably something you shouldn't do because the bms of the console loses its mind, but I had to do it because my only charger for a week was swapping bats with the switch pro controller.
I only figured I should report this to see if it would be easy to fix or even worth fixing but I guess not after all. 
(For completeness' sake, if you hit power in the luma options screen instead of start you can leave the config intact so you don't have to redo your settings, but either way password lock can be circumvented this way for all that's worth.)

profi200, 2023-09-23:
Password/pin locking is not for security and it was not secure in the first place. Deleting Lumas config from the SD card does remove the pin.