LumaTeam/Luma3DS #1954: Random crashes with custom homemenu

cooolgamer, 2023-09-06:
**System model:**

Tested on New 2DS XL and New 3DS XL, it happens to both

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50E SysNAND, also happens on previous versions

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

13.0.2, also happens on previous versions

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4 )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( Clock )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( x )

Enable game patching: ( x )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

By using a custom homemenu with LayeredFS, there is a small chance to randomly crash when you go back to the homemenu from any app/game or applet

**Steps to reproduce:**

1. Use any custom homemenu using LayeredFS

2. Press home on any app/game or applet

3. Be unlucky

**Dump file:**

rdistiso, 2023-10-24:
Try making a new theme, it could be corrupted

cooolgamer, 2023-10-24:
Themes ≠ custom homemenu, they're not the same thing. All my themes works fine, no crash without custom homemenu

rdistiso, 2023-10-24:
> Themes ≠ custom homemenu, they're not the same thing. All my themes works fine, no crash without custom homemenu

Then try rebuilding or update the parts that need to in order to work with 11.17-50
But try asking in homebrew discord

Silver-Volt4, 2024-05-20:
I'm experiencing a similar problem when a modified HOME menu .cia is installed onto the device.
[crash_dump_00000027.dmp](https://github.com/LumaTeam/Luma3DS/files/15376473/crash_dump_00000027.dmp)
It's possible that this is not a Luma bug, but I hardly have the knowledge to say for certain.

korbosoft, 2025-07-15:
This can be easily reproduced by opening and closing the home menu 11 times.

closed 2025-12-12 by TuxSH