LumaTeam/Luma3DS #1955: Various games disconnect my ds from internet when I try to stream

theatrekidhestia, 2023-09-06:
Hi, I was trying to stream Ocarina of Time 3d, but every time I opened it it would disconnect my console from the internet, which stopped the screenshare through Snickerstream. The same thing happened with my pokemon game card, and with the Minish Cap virtual console, but not with Majora's Mask, Pokemon X and the pokemon silver virtual console. there doesnt seem to be any consistancy, except for "if it disconnected the first time it always disconnects it". Is there any way to fix this?

Masamune3210, 2023-09-06:
Gonna go out on a limb and say that you have a o3ds?

theatrekidhestia, 2023-09-06:
> Gonna go out on a limb and say that you have a o3ds?

nope, new.
I kept doing tests and it seems to be only my OoT card, the Minish Cap virtual console, and my 23-in-1 pokemon card, everything else stays connected

That explains why thats the only one of my downloaded ones thadoes this, good to know

Masamune3210, 2023-09-06:
Oot is the weird one here, but I assume it's something the game itself or it's request memory mode does, the other two make sense as I assume the Pokemon card is a DS card. Anything that is DS or GBA runs in its own mini-firmware, Luma can't touch it

theatrekidhestia, 2023-09-06:
Okay, thank you, that, explains why only these few did it. Thanks for your help!

Masamune3210, 2023-09-06:
No problem, please close the issue if everything makes sense now

closed 2023-09-06 by theatrekidhestia

PabloMK7, 2023-09-06:
Hello, for future reference please keep in mind that this place should only be used to report issues about Luma3DS, and not about other homebrew programs.