LumaTeam/Luma3DS #1956: Crash/Arm11 when attempting to load title screen with active plugin

hanagasuki, 2023-09-10:
**System model:**
New 2DS XL
SysNAND version:
11.17.0-50E
Entrypoint:
Boot9Strap
**Luma3DS version:**
v13.0.2
**Luma3DS configuration/options:**
Screen brightness: (4)
Splash: (Off)
Splash duration: (N/A)
PIN lock: (Off)
New 3DS CPU: (Off)

Enable loading external FIRMs and modules: ( )
Enable game patching: (x)
Show NAND or user string in System Settings: (x)
Show GBA boot screen in patched AGB_FIRM: ( )
Set developer UNITINFO: ( )
Disable Arm11 exception handlers: ( )
Enable Rosalina on SAFE_FIRM: ( )

**Explanation of the issue:**
When ACNL (Pre-amiibo EU) attempts to load the title screen with a plugin active (Vapecord 1.9.6) it crashes/throws Arm11.
**Steps to reproduce:**
1. Activate plugin.
2. Boot the game and wait.
[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/12567863/crash_dump_00000000.zip)

Shadow322, 2023-09-10:
> **System model:** New 2DS XL SysNAND version: 11.17.0-50E Entrypoint: Boot9Strap **Luma3DS version:** v13.0.2 **Luma3DS configuration/options:** Screen brightness: (4) Splash: (Off) Splash duration: [...]

Disable plug-in

hanagasuki, 2023-09-10:
I want to play the game with the plugin active though.
If I didn't then I wouldn't have posted this.

PabloMK7, 2023-09-10:
Hello, looks like this issue is related to Vapecord and not Luma3DS. Please contact the Vapecord developers so they can take a look!

closed 2023-09-10 by PabloMK7