LumaTeam/Luma3DS #1957: Sleep issue (Never wakes up)

thekalangatangadvocacy, 2023-09-16:
**System model:**
Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.17.0-50E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
Luma3DS v13.0.2

Enable loading external FIRMs and modules: ( NO)

Enable game patching: ( NO)

Show NAND or user string in System Settings: ( NO)

Show GBA boot screen in patched AGB_FIRM: (NO )

Set developer UNITINFO: ( NO)

Disable Arm11 exception handlers: (NO )

(Apologies if I have missed any info or formatted wrong)

--

**Explanation of the issue:**

When my Old 3DSXL is left in sleep mode, there is a significant change it will not wake up again, despite multiple reboots. The only way I have managed to solve this issue is to remove the battery, hold the power button for 30 seconds and press many more times, then boot without the SD card in to end up at recovery screen. This has happened to me at least twice now.

**Steps to reproduce:**

1.Close the lid without fully turning the console off

2.Wait a while, and open it to find a black screen, that is not resolved until a full battery removal power cycle

PabloMK7, 2023-09-16:
Hello,

The problem reported here seems to be specific to your situation and not a general issue of Luma3DS. You will get better and more fluent support if you ask in a forum related to 3DS homebrew or in the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

closed 2023-09-16 by PabloMK7

Devnol, 2025-05-14:
Having a similar issue with 13.3.2, how do you know this isn't caused by Luma3DS? I've noticed this also occurs on 13.3.2 and only if the console is put to sleep with WiFi turned on.