LumaTeam/Luma3DS #1960: Compile Error：implicit declaration of function

asypost, 2023-09-19:
git clone the latest code and compile,the output says:
D:/Luma3DS/sysmodules/rosalina/source/menus/sysconfig.c:268:9: error: implicit declaration of function 'ACI_LoadNetworkSetting' [-Werror=implicit-function-declaration]
  268 |         ACI_LoadNetworkSetting(i);
      |         ^~~~~~~~~~~~~~~~~~~~~~
D:/Luma3DS/sysmodules/rosalina/source/menus/sysconfig.c:269:9: error: implicit declaration of function 'ACI_GetNetworkWirelessEssidSecuritySsid' [-Werror=implicit-function-declaration]
  269 |         ACI_GetNetworkWirelessEssidSecuritySsid(ssids[i]);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1.exe: all warnings being treated as errors

ihaveamac, 2023-09-19:
Those are newer functions introduced in libctru recently: https://github.com/devkitPro/libctru/commit/5b2a9f6136e5f26b22e8113795732096cb445de8

PabloMK7, 2023-09-19:
You need to manually compile and install the latest libctru. git clone it then make install

closed 2023-09-19 by PabloMK7