LumaTeam/Luma3DS #1964: Error with 2ds xl crashing when opening apps

crimson156, 2023-10-05:
My 2ds xl started crashing when I was playing shovel knight shovel of hope
and now it crashes when I open any apps.
[Crash dumps.zip](https://github.com/LumaTeam/Luma3DS/files/12824868/Crash.dumps.zip)
I am using the newest version if luma

Masamune3210, 2023-10-05:
check your sd card for errors or if its failing. usually when something crashes like that followed by other things immediately crashing, its a sign that the filesystem on the sd card has become corrupted or that the card is about to die

Shadow322, 2023-10-05:
> My 2ds xl started crashing when I was playing shovel knight shovel of hope and now it crashes when I open any apps. [Crash dumps.zip](https://github.com/LumaTeam/Luma3DS/files/12824868/Crash.dumps.zip) I am using the newest version of luma

This not a help forum take it to gbatemp

closed 2023-10-09 by crimson156