LumaTeam/Luma3DS #1967: [Feature request] 3D ON/OFF functionality to Rosalina Menu

MuscleDaddy1985, 2023-10-10:
Many 3DS/3DSLL devices have broken 3D slides. Can you add functionality to Rosalina Menu to turn 3D on or off

MuscleDaddy1985, 2023-10-10:
很多Old3DS/LL的3D滑轨开关出现故障，3DS自带的家长控制功能使用又很不方便，能否给Rosalina Menu增加一个3D ON/OFF的选项，随时开关裸眼3D功能？感激不尽。
Many 3D slide switches of Old3DS/LL have malfunctioned, and the built-in parental control function of 3DS is inconvenient to use. Can you add a 3D ON/OFF option to Rosalina Menu to switch on and off the naked eye 3D function at any time? Thank you very much.

urherenow, 2023-10-10:
Jesus, dude, stop making issues and just fix your slider. Replacements are less than $2. And maybe search the issues here and see what the answer was, because you're not the first person to ask.

rdistiso, 2023-10-24:
> Many 3DS/3DSLL devices have broken 3D slides. Can you add functionality to Rosalina Menu to turn 3D on or off

1. Do you really use 3D?
2. This probably won't be implemented for awhile because I think the system would have to send a power surge saying 'Turn on 3D'