LumaTeam/Luma3DS #1969: [Feature Request] Configurable NTP IP address

BananaBaagel, 2023-10-23:
I'm requesting this because I want to be able to use ntp from a specific NTP server. That being my local firewall which provides network time.

Also, if the request sent out gets rejected by a firewall or does not respond, the 3DS freezes and has to be forcibly shut down (hold power button). In my case, I have an OPNsense firewall that blocks any outbound traffic unless specifically allowed. And that same server also provides NTP. In the logs, it blocked the request out for NTP. If this is something that's actually gonna get fixed then uh I have a new 3ds xl (red), both luma 12.0 and 13.0.2 have the same problem (I updated today to try and fix it). I installed CFW almost a year ago so I don't remember what my installation looked like. I think it involved DSiWare.

kry008, 2023-11-26:
> 3DS freezes and has to be forcibly shut down (hold power button)
I have the same problem, maybe place ip in config.ini will be good choice? Now i have manualy change and compile it to use own IP of NTP host. For manual change you can do it in ntp.h (sysmodules/rosalina/source) @BananaBaagel

eku, 2023-11-26:
> I have an OPNsense firewall

What's wrong with a rule that directs all NTP requests to the OPsense? It's called Destination NAT.

kry008, 2023-11-26:
> > I have an OPNsense firewall
> 
> What's wrong with a rule that directs all NTP requests to the OPNsense? It's called Destination NAT.
> 
> `iptables -A PREROUTING -t nat -p udp --dport ntp -j REDIRECT --to-port 123`

Not in every network you can do that thing, sometimes someone manage it and you cannot access where is blocked default ip from Luma

eku, 2023-11-27:
> > > I have an OPNsense firewall
> Not in every network you can do that thing, sometimes someone manage it and you cannot access where is blocked default ip from Luma

The author of this issue has the option. And that's all I was referring to.