LumaTeam/Luma3DS #1976: [Feature Request] Ability to toggle face tracking on "new" 3ds systems

F3arM3, 2023-12-15:
I wear glasses and that sometimes doesn't play well with the face tracking on the "new" 3ds systems, especially when playing in the dark, so I was wondering if there could be a toggle implemented into Rosalina that allows one to swap to the stable 3d of the "old" systems?

(Sorry for putting this in the luma 3ds section instead of the rosalina thread, it says it has been closed and no-one has posted there in months so I didn't know if this would be seen)

Shadow322, 2023-12-15:
> I wear glasses and that sometimes doesn't play well with the face tracking on the "new" 3ds systems, especially when playing in the dark, so I was wondering if there could be a toggle implemented [...]

Lasik is the solution

cooolgamer, 2023-12-17:
There's literally a setting for that in system settings, other settings, extra stable 3D.
You can turn it off here

PabloMK7, 2024-01-05:
Closing as official OS already supports it

closed 2024-01-05 by PabloMK7