LumaTeam/Luma3DS #1978: ZL/ZR buttons not working

B3ts, 2023-12-20:
In the game IRONFALL Invasion if the option "Enable loading external FIRMs and modules" is enabled on Luma settings cause the ZL/ZR buttons to not work at all in game.

Shadow322, 2023-12-20:
> In the game IRONFALL Invasion if the option "Enable loading external FIRMs and modules" is enabled on Luma settings cause the ZL/ZR buttons to not work at all in game. With it turned off works as intended.

Then don't use that setting then the buttons will work

CycloKid, 2023-12-22:
@Shadow322 i believe they wanted to be able to have it enabled AND be able to use zl/zr. obviously they can disable that but the point of opening the issue was to fix it.

PabloMK7, 2024-01-05:
> In the game IRONFALL Invasion if the option "Enable loading external FIRMs and modules" is enabled on Luma settings cause the ZL/ZR buttons to not work at all in game. With it turned off works as intended.

I doubt the issue is caused by that option. It's possible it's caused by a loaded plugin. Can you please confirm you weren't using any plugins?

B3ts, 2024-01-06:
I wasn't using any 3gx/ntr plugin

TuxSH, 2024-03-30:
Please upgrade rehid if that's what you were using (highly likely considering "if the option "Enable loading external FIRMs and modules" is enabled on Luma settings")

closed 2024-03-30 by TuxSH