LumaTeam/Luma3DS #1981: Build release exits with error 22

impeeza, 2023-12-31:
**Explanation of the issue:**

Building latest version with the command

`make release `

exits with error:

```
built... boot.firm
curl: (22) The requested URL returned error: 404
make: *** [Makefile:43: boot.3dsx] Error 22
```

The problem is produced because the make tries to download  `https://github.com/fincs/new-hbmenu/releases/latest/download/boot.3dsx` but on last release of 3ds hbmenu (2.4.2)
(https://github.com/devkitPro/3ds-hbmenu/releases/tag/v2.4.2) the asset is not boot.3dsx instead 
3ds-hbmenu-2.4.2.zip which contains:

* boot.3dsx
* config\ssl\cacert.pem

I don't know if the makefile should be changed or ask for a new release of 3ds hbmenu with needed files.

Shadow322, 2023-12-31:
> **Explanation of the issue:**
>
> Building latest version with the command
>
> `make release `
>
> exits with error:
>
> ```
> built... boot.firm
> curl: (22) The requested URL returned error: 404
> make: *** [...]

it compiled luma3ds fine just not able to fetch boot.3dsx aka hbmenu just grab it from the current public release

impeeza, 2023-12-31:
Yeah, I did that manually, just letting you to know in case.

TuxSH, 2024-03-30:
Looking

commit 23e478d6 (2024-05-01): Update release archive rules in Makefile

Closes #1981
closed 2024-05-01 by TuxSH