LumaTeam/Luma3DS #1984: Luma3DS Crash in MicroSD Management when shutdown

gamegod333, 2024-01-06:
System Settings -> Data Management -> MicroSD Management in Computer (パソコンでのmicroSDカードの管理。)

if shutdown with Rosalina's shutdown, It's ok. But short press the POWER button to shutdown, luma3ds crash.
My New3ds system is 11.17.0-50J

[crash_dump_00000000.dmp.txt](https://github.com/LumaTeam/Luma3DS/files/13847608/crash_dump_00000000.dmp.txt)

gamegod333, 2024-01-06:
I have tried several New3DS(all jap ver), several SD card，several System version(all jap ver),  several versions of Luma3DS, and all of them have this issue. 
When the New3DS without luma3ds installed, It's ok. so...It's luma3ds's bug, I think.

Masamune3210, 2024-01-06:
Could you actually do what was asked of you in the other issue and fill out the issue template, please? The questions in it are specifically there to help people track down the issue

gamegod333, 2024-01-06:
ok, I will submit to other issue and fill out the issue template, thank you.

closed 2024-01-07 by gamegod333