LumaTeam/Luma3DS #1985: Luma3ds crash when opening homebrew launcher

angrypig555, 2024-01-06:
**System model:**

New 2DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0.50E
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (default)

Splash: (before)

Splash duration: (0 )

PIN lock: ( no)

New 3DS CPU: (arm11 )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (yes )

Enable game patching: (yes )

Show NAND or user string in System Settings: ( no)

Show GBA boot screen in patched AGB_FIRM: ( no)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( no)

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Homebrew launcher crashes when trying to open it.

**Steps to reproduce:**

1.Open homebrew launcher

2.Crash!

**Dump file:**
[crash_dump_00000001.dmp](https://github.com/LumaTeam/Luma3DS/files/13850055/crash_dump_00000001.dmp)

Jhynjhiruu, 2024-02-08:
I've been encountering a similar issue on 13.0.2 where force-shutting down the console by long-pressing the power button sometimes causes a crash when opening the Homebrew Launcher. So far, I've only seen this on consoles with `Hbmenu autoboot` set to `3DS`, where it makes the system unbootable. Resetting `Hbmenu autoboot` to `Off`, then manually launching the homebrew launcher from the system menu clears the problem.

Duckiedec, 2024-03-10:
I had the same problem. All I did was removing boot.3dsx and boot.firm. After that, I downloaded the same version, placed the same files, ran homebrew and it worked.

TuxSH, 2024-03-24:
> I've been encountering a similar issue on 13.0.2 where force-shutting down the console by long-pressing the power button sometimes causes a crash when opening the Homebrew Launcher. So far, I've [...]

Correct, alas I haven't been able to root-cause this. You just need to make sure to press the power button normally, and let the power button screen appear. The workaround you mentioned is the correct one.

Not sure if this applies to OP, though. Maybe they haven't set the correct title for hbmenu (rosalina menu -> misc)

closed 2024-03-24 by TuxSH

angrypig555, 2024-04-20:
I dont have autoboot enabled everything worked fine until a day it didnt. i did not do any modifications.