LumaTeam/Luma3DS #1989: High Gamma Crashes the Console

Bobsican, 2024-01-15:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):** 

11.17.0-50U SysNAND, no EmuNand

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v13.0.2

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

Splash duration: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: (Off)

Enable game patching: (Off)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (On)

Set developer UNITINFO: (Off)

Disable Arm11 exception handlers: (Off)

Enable Rosalina on SAFE_FIRM: (Off)

--

**Explanation of the issue:**

Using the gamma customization feature of the Rosalina menu, it's possible to trigger an exception and thus crash the console, interestingly there does seem to be already a cap set for a minimal gamma value, and thus trying to go for a negative gamma value does not crash the console.

**Steps to reproduce:**

1. Turn on the 3DS normally.

2. On the Home Menu open the Rosalina menu.

3. Go to "Screen filters..."

4. Go to "Advanced configuration"

5. On either screen's section just go to the "Gamma" section and hold right in the D-Pad until it crashes the console, a Gamma value of about 17.0 does the trick, R can be held for faster results.

**Dump file:**
[crash_dump_00000055.zip](https://github.com/LumaTeam/Luma3DS/files/13937882/crash_dump_00000055.zip)

cooolgamer, 2024-02-27:
can reproduce, this also happen to me

TuxSH, 2024-03-22:
Ah, right, that's because `powf` underflows and I don't set thread-local newlib structs correctly, thus it crashes writing errno. I'll just limit the gamma value to 8 or something (values this high don't make sense)

commit 64293f89 (2024-05-01): Config changes and fixes, see description

- Remove allow_updown_leftright_dsi (option was unused and useless)
- Hide enable_dsi_external_filter. Do not mention out-of-date wiki,
  fixes #1992
- Limit gamma to a more reasonable value of 8, fixes #1989
closed 2024-05-01 by TuxSH