LumaTeam/Luma3DS #1993: [Feature Request] Add LayeredFS support for updates

CyberYoshi64, 2024-01-27:
LayeredFS currently only allows patching the base game romfs because it only replaces files for the mountpoint `rom:`.
Because of this update data (using the `ro2:` mountpoint) are left untouched and the game prioritizing the update over the base romfs simply breaks LayeredFS changes. I have eluded to this before but I wasn't great at describing it back then (Issue 1659).
The Mario Kart 7 1.2 update is a recent example. As a result, mods that change the in-game text (or fonts) simply don't work anymore, because the game reads the files from the update and thus not from LayeredFS either.

It would be appreciated if updates could obtain LayeredFS support. Perhaps by adding a new folder called `update-romfs` or so.
Thank you for your consideration.

AuroraWright, 2024-01-29:
it’s already supported, but maybe it doesn’t support the mount name in mario kart 7 (?)
see here for already supported mount names https://github.com/LumaTeam/Luma3DS/blob/e2778a45cf379270c3039aab16bb607178561cc1/sysmodules/loader/source/patcher.c#L582

CyberYoshi64, 2024-01-29:
It's `ro2`, not `rom2`

PabloMK7, 2024-02-01:
> There is `rom2` but some games use `ro2` instead. However, MK7 seems to use another name that is a little hard to find. I'll try experimenting with this name list.
> 
> Mario Kart 7 mounts it in a somewhat unusual way. I found the string `pat1` in the code, but I have to leave out the colon in that string array, otherwise LayeredFS doesn't apply to the update.

If you PR that simple fix can be merged

closed 2024-02-02 by CyberYoshi64