LumaTeam/Luma3DS #2005: [REQUEST] Hue shift filter

Widdiful, 2024-03-16:
Would it be possible to add a hue shift feature to the screen filters? I'm colourblind and I find that it can help a lot. Puzzle games in particular are very difficult as it's hard or even just impossible to tell some colours apart, but if I can shift them to something else then many games become easily playable like normal. I think this feature could be very useful to people who are colourblind like me, without having to make per-game mods.

TuxSH, 2024-03-17:
Hi, I'm not sure this is possible. What the "screen filters" do is change the **per-channel** pixel brightness values (basically framebuffer pixel values go through the LUT to obtain the value used by the LCD). The fact that it is per-channel is an hardware limitation

profi200, 2024-03-17:
I don't think hue is possible with the hardware 1D lookup table but shifting colors is possible. It's already used for redshift.

closed 2024-10-19 by TuxSH