LumaTeam/Luma3DS #2006: luma3ds dosnt  let 3ds boot while  ds cart is inserted

GweshaDeneb, 2024-03-19:
System model: New 3DS XL

**SysNAND version  11.17.0-50U

Entrypoint:Boot9Strap

Luma3DS version: 13.0.2

Explanation of the issue:
 n3dsxl(might be the same for other 3ds versions) does not  fully boot up  with physical ds  game inserted into gamecard slot( stays in black screen) blue light stays but no yellow light and screen lighting up indicating it started up properly
booting the select options work  on start up but save/exiting dosnt help either stays in black screen after still 

also if it matters my ds cart is one of the other games (not pokemon ) that have the infrared sensor and i dont have a normal cart to try with it instead.

**Steps to reproduce:**

have ds gamecard inserted on boot up

PabloMK7, 2024-03-21:
I've seen this happen in the past if the DS card was not properly inserted. I guess you already tried to remove it and insert it again?
Which is the game specifically?

GweshaDeneb, 2024-03-22:
> I've seen this happen in the past if the DS card was not properly inserted. I guess you already tried to remove it and insert it again? Which is the game specifically?

i tried with a copy of walk with me  for the ir function but i realized after trying to reboot my n3ds it wouldn't boot unless i took out the ds cart  ive tryed it several times so i know it was because of having it inserted not sure why it does it though..

rdistiso, 2024-03-29:
Have you tried booting without it inserted and then once booted inserting the game card?