LumaTeam/Luma3DS #2011: [Feature Request] C-Stick emulation for O3DS in InputRedirection

isa56, 2024-04-02:
Is it possible to emulate the Circle Pad Pro / C Stick on the Old 3DS using Input Redirection?  That would be a very interesting feature for FPS or camera control on O3DS.

TuxSH, 2024-04-24:
No, it is not possible to do that, this is far from trivial and requires fully reimplementing the `ir` sysmodule (AFAIK), thus this will not be done in the near future.

closed 2024-04-24 by TuxSH

isa56, 2024-04-25:
@TuxSH thank you for your answer. That's a pity, but totally understandable.

Zackeroo, 2025-07-11:
Yea me too haha . Them calling it trivial is sad. I wanted to use my Xbox controller for an FPS game on the 3ds. Everything was working fine but I couldn't move the camera around. The game had a circle pad pro option so I was hoping to use the right stick on the Xbox controller for that.