LumaTeam/Luma3DS #2012: Excuse me,why can't luma show too many project cheats

Rendigga, 2024-04-03:
LUMA13.02   new3dsll   Boot9Strap  11.17 
Explanation of the issue:
Only about 100 items can be displayed,but my monster hunter XX has 300 cheat items.I hope to have it show more cheat items in the next release

Rendigga, 2024-04-03:
[image]

Rendigga, 2024-04-03:
[image]

Rendigga, 2024-04-03:
Is there a problem with my luma version or settings, or is it that luma is not implemented at the moment😭😭😭

Rendigga, 2024-04-03:
[0004000000197100.txt](https://github.com/LumaTeam/Luma3DS/files/14851362/0004000000197100.txt)

Please try it

PabloMK7, 2024-04-24:
Hello, there is a limit with how many cheats can be loaded at a time due to memory and execution time restrictions.

closed 2024-04-24 by PabloMK7

Rendigga, 2024-04-24:
Is there a chance to break through this restriction in the next version

Rendigga, 2024-04-24:
> Hello, there is a limit with how many cheats can be loaded at a time due to memory and execution time restrictions.

Is there a chance to break through this restriction in the next version？🥹 Or can we make luma display more

PabloMK7, 2024-04-24:
Yes, by downloading more ram to your console.

Rendigga, 2024-04-24:
> 是的，通过将更多内存下载到您的控制台。

thank！🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷

Rendigga, 2024-04-24:
> Yes, by downloading more ram to your console.

thank！🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷🌷

Masamune3210, 2024-04-25:
........

Rendigga, 2024-05-07:
> 是的，通过将更多内存下载到您的控制台。

Hello, I saw that luma was recently updated. Has the memory limit problem been solved? can more cheats be displayed now?pleases🥺🥺🥺

Rendigga, 2024-05-07:
@PabloMK7