LumaTeam/Luma3DS #2014: Luma3DS v13.02 [HID crashes on boot]

Rodostlyn69, 2024-04-08:
When booting into 3DS with Luma3DS v13.02, Luma3DS crashes and shows an Arm11 data abort by hid process (0004013000001D02). It does that after enabling both "Enable loading external FIRMs and modules" and "Enable game patching" options. Crash dump file:
[crash_dump_00000025.dmp](https://github.com/LumaTeam/Luma3DS/files/14904149/crash_dump_00000025.dmp)

Superdude11235, 2024-04-09:
I've been having that same issue, though only with "enable loading external FIRMs and modules. I hope it gets fixed so I can use Pretendo.

urherenow, 2024-04-09:
And do you have the issue with external modules enabled, but with NO external modules to load? If not, then your issue doesn’t belong here. It belongs with that module.

Rodostlyn69, 2024-04-09:
> And do you have the issue with external modules enabled, but with NO external modules to load? If not, then your issue doesn’t belong here. It belongs with that module.

I have the sysmodules, unlikely, so thats not the problem.

urherenow, 2024-04-09:
I meant, leave the option on, but don’t use those sysmodules. When you add a particular one and get the error, then that module probably needs an update.

Rodostlyn69, 2024-04-09:
> I meant, leave the option on, but don’t use those sysmodules. When you add a particular one and get the error, then that module probably needs an update.

I tried removing the sysmodules, but that didn't changed anything at all, same error.

Superdude11235, 2024-04-09:
I solved the issue by updating rehid. Try doing that and see if that helps! Just download the latest version of rehid helper and update it through there.

closed 2024-04-09 by Rodostlyn69

Superdude11235, 2024-04-09:
Glad to have helped!