LumaTeam/Luma3DS #2016: 3ds boots to luma menu, Freezes completely freezes when I save & exit

groovyrobloks2, 2024-04-15:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS] Old 3DS

**SysNAND version (+emu/redNAND version if applicable):** Sys 11.17.0-50U

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):** just the power button 

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:** 13.0.2

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: ( after)

Splash duration: ( )

PIN lock: ( off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
When I boot my old 3DS it immediately opens Luma3DS config, when I go to save and exit it freezes and requires me to hold the power button to do anything. I've been able to bypass it sometimes by holding down A+L+R+Up but I really shouldn't have to do that just to use my console.

**Steps to reproduce:**

1. Turn on 3ds

2. Get stuck in luma config 

**Dump file:**

Masamune3210, 2024-04-15:
If you have Luma on your NAND, try taking your SD card out. Does it still do it? If no, SD card might be corrupted or going out.

groovyrobloks2, 2024-04-15:
One of the first things I tried, I pulled out my SD card and it still boots to luma and freezes when I try tk save and exit.

rdistiso, 2024-04-16:
Try an SD clean: https://wiki.hacks.guide/wiki/SD_Clean/Windows

rdistiso, 2024-04-23:
And also update your Sysnand

closed 2024-04-24 by PabloMK7