LumaTeam/Luma3DS #2022: Streetpass Mii Plaza doesn't work on a 64GB SD card

xtomasnemec, 2024-05-04:
**System model:New 3DS**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):11.17.0-50E EmuNAND**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS)Boot9Strap:**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:13.1**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (After)

Splash duration: (3s)

PIN lock: (none)

New 3DS CPU: (disabled)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (Yes)

Enable game patching: (Yes)

Show NAND or user string in System Settings: (Yes)

Show GBA boot screen in patched AGB_FIRM: (Yes)

Set developer UNITINFO: (No)

Disable Arm11 exception handlers: (No)

Enable Rosalina on SAFE_FIRM: (No)

--

**Explanation of the issue: Streetpass mii plaza 5.1 doesn't launch (hangs on the splash screen)**
[image]

**Steps to reproduce:**

1. Install Streetpass Miiplaza 5.1 on a 64gb sd card

2. Launch it

**Dump file:**

xtomasnemec, 2024-05-05:
> You didnt format your sd card correctly it runs fine on my 258GB sd card

I have formated it to FAT32 32kb MBR

PabloMK7, 2024-05-05:
Hello, this is probably caused by a corrupted SD and it's not related to a Luma3DS bug. In any case, to get further assistance please ask in a homebrew forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2024-05-05 by PabloMK7

xtomasnemec, 2024-05-06:
> Hello, this is probably caused by a corrupted SD and it's not related to a Luma3DS bug. In any case, to get further assistance please ask in a homebrew forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

Reformating didn't help

Masamune3210, 2024-05-06:
You missed the second part of that message, didn't you.....

xtomasnemec, 2024-05-12:
Fixed it. Pretendo sysmodules were corrupted