LumaTeam/Luma3DS #2025: New 3ds Crashes when trying to play video with Thirdtube

doog-aan, 2024-05-08:
**System model:**
New 3DS XL
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.13.0-45E EmuNAND
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
 v10.2 stable
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
When I try to load a youtube video using the homebrew app Thirdtube the system crashes when i try to play any video

**Steps to reproduce:**

1. Go onto Thirdtube and try to play a video

2. 

**Dump file:**

[crash_dump_00000007.dmp](https://github.com/LumaTeam/Luma3DS/files/15254835/crash_dump_00000007.dmp)

urherenow, 2024-05-09:
So what? This isn’t the Thirdtube repo.

PabloMK7, 2024-05-09:
Hello, this repo is for tracking issues of Luma3DS, and not for other homebrew applications.

closed 2024-05-09 by PabloMK7

InaBuild, 2024-05-10:
bruh