LumaTeam/Luma3DS #2034: (Regression) When upgrading, Luma will not boot from any NAND if SysNAND CTRNAND cannot be mounted

aspargas2, 2024-05-22:
**System model:** New 2DS XL (but will happen on all)

**SysNAND version (+emu/redNAND version if applicable):** EmuNAND 11.15.0-47J (but will happen on all supporting Emu)

**Entrypoint (How/what you're using to boot Luma3DS):** Boot9Strap

**Luma3DS version:** v13.1.1 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: (x)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**

Due to the upgrade process involving an attempt to copy boot.firm to SysNAND CTRNAND, the latest version of Luma cannot boot into any EmuNAND on a system without a mountable SysNAND CTRNAND until a version of Luma which implements the latest config version has been successfully booted at least once (so a latest config is in MCU regs and the upgrade process no longer tries to happen). If this is attempted, Luma will give the `f_mount returned <something>` error because it tried to mount SysNAND CTRNAND. This is particularly annoying in the case of a system with a totally nonfunctional SysNAND that has to be permanently tied to an EmuNAND and an ntrboot card.

**Steps to reproduce:** (assuming a system that has never boot a Luma version implementing latest config version)

1. Partition the SD card to have an EmuNAND, and setup this EmuNAND such that it is bootable
2. Inject some garbage data into SysNAND CTRNAND such that it can no longer be mounted
3. Enable `Autoboot EmuNAND` in Luma config (at this point, old versions of Luma can boot the system fine)
4. Attempt to boot the latest version of Luma
5. The error `f_mount returned <something>` will appear

TuxSH, 2024-05-22:
Does this happen on v11.0 as well?

Veund, 2024-05-22:
aspargas2 has been helping with my 3ds, and this only happens on v13 and up, it is fine on 12.0.1 and below

TuxSH, 2024-05-25:
Could you try this file please? [boot.zip](https://github.com/LumaTeam/Luma3DS/files/15444112/boot.zip)

closed 2024-05-25 by TuxSH

Veund, 2024-05-25:
Yes this works

commit 41dd8bd2 (2024-05-25): Fix #2034