LumaTeam/Luma3DS #2036: [Feature Request] Better brightness control

TrixieUA, 2024-05-25:
Allow lower brightness and more maximum brightness(aka 6 level brightness).

profi200, 2024-05-25:
Lower brightness is not really an issue but higher increases the risk of hardware damage. We don't know the exact ratings for the backlights and power supplies they use. They already limit the current in the backlight curves.

If you plug in the charger and set the brightness to maximum in HOME menu you will already get a boosted brightness if the current limit allows it. This is something not very well known.

TuxSH, 2024-05-26:
Current implementation uses gsp which clamps the brightness values and I do not ever intend to add an option to go outside the calibration.

closed 2024-05-26 by TuxSH