LumaTeam/Luma3DS #2037: System crashes when opening a menu (data abort exception)

Luke1002, 2024-05-26:
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

11.17.0-50E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v13.1.1

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Every time I start Fantasy Life (european cartridge, same region as my console), the menus start glitching and the console crashes if I try to open the Challenges Menu from within the Licences Menu after a random period of time that varies between immediately opening the game and 10-15 min from when I start playing.

**Steps to reproduce:**

1. Start Fantasy Life
2. Load a save
4. Open the Menu
5. Open the Licences submenu
6. Open the Challenges menu

**Dump file:**
[crash_dump_00000006.zip](https://github.com/LumaTeam/Luma3DS/files/15447148/crash_dump_00000006.zip)

PabloMK7, 2024-06-02:
Hello, this is not a Luma3DS bug as that game is confirmed to work. You may need locale emulation in some games. Or the cartridge may be faulty.

closed 2024-06-02 by PabloMK7

profi200, 2024-06-02:
You may wanna check the game using this. Make sure to let the author know if your game was corrupted and to which extend it got fixed (or not).

https://github.com/skawo/GodMode9-with-Cartridge-Fixer

Luke1002, 2024-07-22:
> Hello, this is not a Luma3DS bug as that game is confirmed to work. You may need locale emulation in some games. Or the cartridge may be faulty.

Sorry for the late reply, had lots of things happening at once and didn't have time.
I don't know which locale emulation I should try for, my console and my game are from the same region. As for the cartridge, I checked using the GodMode9 Verify function and it completed successfully. Any help in identifying the issue?