LumaTeam/Luma3DS #2038: New3ds input redirection cause console crash:

DragonautMk, 2024-05-28:
New3ds input redirection cause console crash:
I'm on latest luma, i've booted my switch in to android and installed input redirection app. Some times it lets me play a bit, sometimes it just crashes. It's a luma bug? It's a redirection app bug?

urherenow, 2024-05-28:
What are you even talking about, booting a 3DS into Android? Whether that's ACTUALLY what you're doing, or if you're running a 3DS emulator on Android,I don't know what that would have to do with Luma3ds. The 3DS doesn't even have enough ram to properly support Android OS, much less trying to run a redirection app on top of it.

DragonautMk, 2024-05-28:
No, maybe i've explained myself wrong.
I booted my nintendo switch in to android to install input redirection apk. This lead me to control my 3ds using my switch.