LumaTeam/Luma3DS #2040: luma won't update

Kirbsers, 2024-05-30:
**System model:** Old 3ds

**SysNAND version:** 11.17.0-50E

**Entrypoint:** Boot9Strap

**Luma3DS version:** 13.0.2

**Luma3DS configuration/options:**

Screen brightness: (4 )

Splash: (Off )

PIN lock: (Off )

Enable loading external FIRMs and modules: (Off )

Enable game patching: (On )

Show NAND or user string in System Settings: (On )

Show GBA boot screen in patched AGB_FIRM: (On )

Disable Arm11 exception handlers: (Off )

--

**Explanation of the issue:** I saw a video today that said there was an update for luma and the homebrew channel and right after i went on to update both of the files on my sd card with universal updater. i restarted my 3ds and checked if the version updated to luma 13.1.1 but it was stuck on version 13.0.2 despite me having updated it. i turned it back off and put the sd card into my pc and tried updating it with the github files, but still nothing has changed and its still stuck on version 13.0.2. strangely enough the homebrew version updated just fine.

**Steps to reproduce:**

1. be on version 13.0.2 of luma

2. update per universal updater

3. restart and check version

4. still on version 13.0.2

5. remove sd card to put it into pc

6. manually update and put the sd card back into the 3ds

7. still on version 13.0.2 despite trying two ways to update luma

PabloMK7, 2024-05-30:
Hello, your SD card is corrupted. Copy all the files from your SD card to your computer, format the card to FAT32 and finally copy everything back. If the problem still persists please ask for help in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2024-05-30 by PabloMK7

urherenow, 2024-05-31:
meh. or use godmode9 to delete Luma from your ctrnand. 

Maybe the card is corrupt, and maybe it would be more obvious if Luma didn't revert to its ctrnand copy and happily load anyway, but didn't having it in ctrnand used to be a choice? It really should still be a choice.

Masamune3210, 2024-05-31:
Horse as already been beaten to a paste at this point. For what its worth, its completely normal and actually quite common for fallbacks to not be a user-side choice. The 3DS, for example, with its FIRM0/FIRM1 set up.

PabloMK7, 2024-06-01:
> meh. or use godmode9 to delete Luma from your ctrnand.

Definitely don't do this, as the console won't boot then at all. The file boots from ctrnand for a reason.

urherenow, 2024-06-01:
what a strange reply coming from you. If it wouldn't boot, then how did it get there in the first place? It doesn't exist there the first time you use it.

PabloMK7, 2024-06-01:
> what a strange reply coming from you. If it wouldn't boot, then how did it get there in the first place? It doesn't exist there the first time you use it.

What? What does that have to do with anything? Anyways this conversation won't go anywhere, the user has already gotten the help they needed.