LumaTeam/Luma3DS #2044: Compiling problems

Golem642, 2024-06-02:
I cloned this git repo, but when i try to make there's apparently some errors into main.c

[image]

I have everything installed (git, libctru, makerom and firmtool, althought this doesn't seem to come from any of these), is this normal/ am i missing something or am i not the only one having this problem ?

Thanks

PabloMK7, 2024-06-02:
Hello, you are using an outdated libctru. Make sure you are using the latest version provided by devkitpro.

closed 2024-06-02 by PabloMK7

Golem642, 2024-06-02:
Sorry, I'm new to pacman. I thought running `pacman -Syu` would update everything but it seems I was mistaken and after running the command again with libctru at the end, it fixed it. Thanks