LumaTeam/Luma3DS #2054: An exception has occurred Arm 11 (core 0)

lilmike14, 2024-07-03:
**System model:**
Old 2ds

**SysNAND version (+emu/redNAND version if applicable):**

I'm afraid I can't find out, I get an error on boot

**Entrypoint (How/what you're using to boot Luma3DS):**
Luma says "Booted from SD via B9S"
Boot9Strap

**Luma3DS version:**

13.1.2

**Luma3DS configuration/options:**

I'll delete the options the configuration screen doesn't show me.

Screen brightness: (4 )

Splash: ( Off)

PIN lock: ( off)

Hbmenu autoboot: (off)

Enable loading external FIRMs and modules: (off )

Enable game patching: (off )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: (off )

--

**Explanation of the issue:**

DS boots up with a black screen showing "an exception has occurred"
Arm 11 (core 0)
data abort
Translation - Section
Menu (0004003000009802)

Last night I decided to use the 2ds after many years.

I installed universal-updater vie FBI QR code, then Twighlight Menu++

From there I updated luma to 13.1.2 and installed a couple of games via FBI QR code.

All was good last night, I played some new games, turn the ds on and off over the course of a few hours and had no issues.

The same could be said for this morning, but when I turned it on in the afternoon I got the error.

So far I've tried:
Going down in luma version as far as 9.1
Reformatted the SD (32gb) as FAT32, 32kb cluster size. Then did a write/verify test with H2testw. The test didn't have any errors.
I've also tried deleting the 00000098 file in Nintendo 3ds/.../..../extdata/00000000

**Steps to reproduce:**

1.Turn on 2ds

2.

**Dump file:**

[crash_dump_00000000.zip](https://github.com/user-attachments/files/16075881/crash_dump_00000000.zip)

TuxSH, 2024-07-06:
What happens if you boot w/o SD card?

gio46464, 2024-08-19:
Same thing happens to me a lotq

TuxSH, 2024-10-02:
No info given, closing

closed 2024-10-02 by TuxSH