LumaTeam/Luma3DS #2056: Time offset issue status

mariomadproductions, 2024-07-11:
The release notes for [v10.3](https://github.com/LumaTeam/Luma3DS/releases/tag/v10.3) say "Programs like ctr-no-timeoffset should not be needed anymore. Also, even if 3ds.hacks.guide recommends it and GodMode9 mandates it, time offset nullification should not be done".

However I asked on the Nintendo Homebrew discord, and they suggested this is no longer the case, and you should nullify the time offset when GodMode9 asks.

Is this correct advice? And if so, could it be documented in the readme/wiki and an addendum be added to those release notes?

Thanks.

mariomadproductions, 2024-07-14:
I guess this advice did change, since the rosalina menu apparently has a "nullify user time offset" option now...

impeeza, 2024-07-14:
Latest Luma have this menu under «Miscellaneous options menu»

[image]

you can sync your RTC from a NTP source and remove the User Time Offset.

TuxSH, 2024-07-15:
Question has been answered.

GM9 requests time offset nullification be done, but this is something that GM9 should ideally not rely on.

closed 2024-07-15 by TuxSH