LumaTeam/Luma3DS #2060: issue when trying to load up old 2ds with luma 13.1.2

KatsumaCodes, 2024-07-21:
Old 2ds

11.17.0-50u  (+emu/redNAND version if applicable)

Entrypoint (How/what you're using to boot Luma3DS)

Boot9Strap

Luma3DS version: v13.1.2

Luma3DS configuration/options:

Default EmuNAND: (I dont know what it is sorry)

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

Hbmenu autoboot: (off)

--

Autoboot EmuNAND: ( )

Enable loading external FIRMs and modules: (off)

Enable game patching: (off)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (off)

--

Explanation of the issue:
so I saw changing my themes on friday after midnight right after playing
mario party: the top 100
and it gave me the error screen and I went to my dad worried I got hard bricked
my mom and dad tried to help but no luck I read on reddit for a bit and tried booting up on luma and saving hoping 
it would work but you can geuss why I'm here :') I tried to update it with recovery mode but my console is up to date
(l button, r button up on the dpad and the a button I think)

**Steps to reproduce:**

1. power on 

2. thats it :P

Dump file (if applicable):
[crash_dump_00000010.dmp](https://github.com/user-attachments/files/16323210/crash_dump_00000010.dmp)

urherenow, 2024-07-21:
I don’t understand what you’re trying to say. You think a theme bricked your 3DS? This space is to report Luma bugs, not for support. You should search reddit or gbatemp or even discord for help. If it is a theme, then:

If you get a 'semi' brick from any theme..

On sysnand turn off 3ds, remove sd, turn back on and install a default theme

On emunand browse sd on a pc. Go to Nintendo 3ds - (id0 of your emunand) - (id1) - extdata/ delete the appropriate folder based on your region.

JPN - 000002cc
USA - 000002cd
EUR - 000002ce

Reinsert sd and power on 3ds and install a default theme.

PabloMK7, 2024-07-21:
Hello, this page is for reporting issues for Luma3DS and not for giving assistance. Please ask for help in a homebrew related forum or in the Nintendo homebrew discord server: https://discord.gg/C29hYvh

closed 2024-07-21 by PabloMK7

KatsumaCodes, 2024-07-21:
I'm so sorry I forgot to put that it was a arm11 error I had been trying so hard to get it work since 3am yesterday my bad