LumaTeam/Luma3DS #2067: Luma 3DS Screenshot Feature not working, Operation Failed (0xc8804478)!

BlueyBooey1278, 2024-08-14:
I've been using Luma 3DS for quite a while on my Homebrewed 3DS and have used It from time to time to take screenshots for some of the Games I've been playing, but suddenly the feature stopped working as everytime I try to use the Screenshot feature It gives the same message "Operation Faile (0xc8804478)." and Idk how to fix It, I have tried updating Luma3ds via the Universal Updater Homebrew App but that hasn't fixed the Issue and Im not sure what else to do now so I am asking on here for help!

System model:
[Old 3DS XL]

SysNAND version (+emu/redNAND version if applicable):
[[SYS 11.17.0-50U]

Entrypoint (How/what you're using to boot Luma3DS):
[Boot9Strap]

Luma3DS version:
[v13.1.2]

impeeza, 2024-08-14:
This is not a support forum, please go to Discord or GBATemp.

In this case that error is related to a corrupted SD Card,  please backup all files on your card, Format it FAT32 and test all card using H2Testw if the card is defective change for a new one.

https://gbatemp.net/threads/my-ams-emulator-cant-run-games.615664/post-9885418

Restore the backup of your files.

https://wiki.hacks.guide/wiki/Formatting_an_SD_card/Windows#tabber-32GB_or_smaller

please close this issue because is not a Luma Bug.

BlueyBooey1278, 2024-08-14:
Ahh well okay, thanks for the explaination, I'll probably see If I can find any alternative means of screenshotting games on the 3DS that aren't Luma In the meantime

PabloMK7, 2024-08-14:
Hello, I recommend doing the steps mentioned by impeeza, as corruption may spread to other files. I also recommend joining the Nintendo homebrew discord server for further support.

closed 2024-08-14 by PabloMK7

BlueyBooey1278, 2024-08-14:
I checked the SD Card using both H2Testw like they had said and I even used Windows built In Error Checking tool, both came up with no errors, I am rn looking at other solutions and will probably go check out the Nintendo Homebrew Discord like suggested If I cannot figure out anything else on Reddit or GBATemp!

BlueyBooey1278, 2024-08-14:
> Hello, I recommend doing the steps mentioned by impeeza, as corruption may spread to other files. I also recommend joining the Nintendo homebrew discord server for further support.

I checked the SD Card using both H2Testw like they had said and I even used Windows built In Error Checking tool, both came up with no errors, I am rn looking at other solutions and will probably go check out the Nintendo Homebrew Discord like suggested If I cannot figure out anything else on Reddit or GBATemp!