LumaTeam/Luma3DS #2068: Sometimes when screenshots are taken in they are duped

Trickiy, 2024-08-17:
In the current version of luma sometimes when I take a screenshot the bottom and top screens will be the same:
[image]
[image]
in the game editor the select button is used to swap bottom and top screens. 

Step 1 find a program that is suseptible Thextech works
Step 2 open luma and take a screen shot not 100% garenteed

TuxSH, 2024-08-17:
Does this happen if you remap luma menu combo to something else that doesn't involve the SELECT button?

Trickiy, 2024-08-18:
No it dose not happen.

TuxSH, 2024-10-02:
I've thought about it a bit, but I think I'll close the issue -- what happens is that the game may react to the Rosalina combo. This is very hard to fix properly (requires reimplementing large parts of the system). Workaround is to change the menu combo as mentioned earlier.

closed 2024-10-02 by TuxSH