LumaTeam/Luma3DS #2070: [Feature Request] option to disable sleep mode switch on 2ds / disable sleepmode by closing lid 3ds

GreatNateDev, 2024-08-20:
I have a 2ds that has a broken sleep mode switch that keeps toggle ing on and it would be nice if as a temporary fix i could disable that

urherenow, 2024-08-20:
Turn it off. Soak a Q-tip in IPA and rub it along the switch's groove, and manually toggle the switch on and off real fast. Maybe 20-30 times. Apply more IPA if it dries too quickly. When you're done, wait a good 5-10 minutes for the IPA to evaporate, then turn on your console and try the switch out. You may find that it works much better, if not like new.

impeeza, 2024-08-20:
Sorry is a hardware interruption, no too much to do, you will need to fix the button.

GreatNateDev, 2024-08-22:
> Sorry is a hardware interruption, no too much to do, you will need to fix the button.

bummer

GreatNateDev, 2024-08-22:
> Turn it off. Soak a Q-tip in IPA and rub it along the switch's groove, and manually toggle the switch on and off real fast. Maybe 20-30 times. Apply more IPA if it dries too quickly. When you're [...]

can i do this isopropyl alcohol?

urherenow, 2024-08-22:
That's what IPA is...

closed 2024-08-23 by GreatNateDev